sekigon-gonnoc/qmk_firmware #28: RGB backlighting doesn't work without USB

joric, 2020-02-24:
Can you please help enabling RGB on battery power (I have a large Li-Po and enough current to keep it running for a while)? Looks like RGB stops working without USB but for gods sake I can't find the culprit, rgblight_task seems to be called normally and there are no RGB disabling calls in the usbd.c. Video:

https://youtu.be/H9Vw5rEFeG4

[image]

joric, 2020-05-05:
OK I got it! How could I overlook it? It was an explicit condition in ws2812 driver:
```patch
diff --git a/drivers/nrf52/ws2812.c b/drivers/nrf52/ws2812.c
index 048f020ff..57b5f8ec5 100644
--- a/drivers/nrf52/ws2812.c
+++ b/drivers/nrf52/ws2812.c
@@ -38,10 +38,7 @@ void pwm_handler(nrfx_pwm_evt_type_t event_type) {
 }
 
 void ws2812_setleds(LED_TYPE *ledarray, uint16_t number_of_leds) {
-  if (nrfx_power_usbstatus_get() == NRFX_POWER_USB_STATE_CONNECTED ||
-      nrfx_power_usbstatus_get() == NRFX_POWER_USB_STATE_READY) {
-    ws2812_setleds_pin(ledarray, number_of_leds, RGB_DI_PIN);
-  }
+  ws2812_setleds_pin(ledarray, number_of_leds, RGB_DI_PIN);
 }
 
 void ws2812_setleds_pin (LED_TYPE *ledarray, uint16_t number_of_leds,uint8_t pinmask){
```

RGB sync doesn't work just yet and there's no eeprom support but that thing above was the worst.

elfmimi, 2020-05-05:
How could you overlook it?

closed 2020-05-05 by joric