sekigon-gonnoc/qmk_firmware #43: RGB Matrix with WS2812 on nrf52 branch

Tobiassss, 2021-05-14:
## Feature Request Type

- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)

## Description

The nrf52 branch contains outdated RGB Matrix functionality. Since I have WS2812 LEDs behind each key, it isn't compatible. It only supports IS31xxxx LED Drivers.
 
I'm using a nRFMicro, so I can't switch to the main QMK branch.
Any updates planned to bring the current RGB Matrix implementation also to the nrf52?

joric, 2021-05-15:
How's that? I used WS2812 just fine. https://github.com/joric/nrfmicro/wiki/QMK

Tobiassss, 2021-05-15:
Aren't you using the "RGB Lighting" function of QMK (https://beta.docs.qmk.fm/using-qmk/hardware-features/lighting/feature_rgblight)?
 rules.mk:
`RGBLIGHT_ENABLE = yes`
Which is used for Keyboard Underglow etc.

I mean the "RGB Matrix" function (https://beta.docs.qmk.fm/using-qmk/hardware-features/lighting/feature_rgb_matrix): 
rules.mk:
`RGB_MATRIX_ENABLE = yes`
`RGB_MATRIX_DRIVER = WS2812`
\+ Key - LED mapping over `led_config_t g_led_config =` in keyboard.c
Which is meant for lighting specific, single Keys.

As long as you only do "general" lighting like rainbow effect etc., both functions work fine.
But I want to use key specific "reactive" effects (`enum rgb_matrix_effects`) e.g. you press one key and this key will light up.