sekigon-gonnoc/qmk_firmware #45: [Bug] Compile error when RGBLIGHT_ENABLE = no

tadakado, 2021-07-31:
Compile error when I set RGBLIGHT_ENABLE = no in rules.mk
Here is my quick patch. Thanks,

```
diff --git a/tmk_core/protocol/nrf/bmp.c b/tmk_core/protocol/nrf/bmp.c
index 71ef1d45..8f182d74 100644
--- a/tmk_core/protocol/nrf/bmp.c
+++ b/tmk_core/protocol/nrf/bmp.c
@@ -287,10 +287,12 @@ MATRIX_LOOP_END:
 
 bmp_error_t nus_rcv_callback(const uint8_t* dat, uint32_t len)
 {
+#ifdef RGBLIGHT_ENABLE
     if (len == sizeof(rgblight_syncinfo_t))
     {
         rgblight_update_sync((rgblight_syncinfo_t*)dat, false);
     }
+#endif
     return BMP_OK;
 }
 
diff --git a/tmk_core/protocol/nrf/main_master.c b/tmk_core/protocol/nrf/main_master.c
index 97a1da77..1b206a04 100644
--- a/tmk_core/protocol/nrf/main_master.c
+++ b/tmk_core/protocol/nrf/main_master.c
@@ -85,7 +85,9 @@ host_driver_t nrf_ble_driver = {
   send_system,
   send_consumer
 };
+#ifdef RGBLIGHT_ENABLE
 rgblight_syncinfo_t rgblight_sync;
+#endif
 
 #ifdef VIRTSER_ENABLE
 void virtser_task(void);
@@ -225,8 +227,10 @@ int main(void) {
 
   const bmp_api_config_t * config = BMPAPI->app.get_config();
 
+#ifdef RGBLIGHT_ENABLE
   rgblight_set_clipping_range(0, config->led.num);
   rgblight_set_effect_range(0, config->led.num);
+#endif
 
   bmp_indicator_init(config->reserved[1]);
 
diff --git a/tmk_core/protocol/nrf/sdk15/cli.c b/tmk_core/protocol/nrf/sdk15/cli.c
index 929fc528..82be36e2 100644
--- a/tmk_core/protocol/nrf/sdk15/cli.c
+++ b/tmk_core/protocol/nrf/sdk15/cli.c
@@ -10,7 +10,9 @@
 #include "debug.h"
 #include "print.h"
 #include "quantum.h"
+#ifdef RGBLIGHT_ENABLE
 #include "rgblight.h"
+#endif
 #include "eeprom.h"
 
 #include "bmp.h"
@@ -437,12 +439,14 @@ static MSCMD_USER_RESULT usrcmd_led_control(MSOPT *msopt, MSCMD_USER_OBJECT usro
         msopt_get_argv(msopt, 1, arg, sizeof(arg));
         uint32_t mode = (uint32_t)atoi(arg);
         dprintf("set led mode:%d\r\n", mode);
+#ifdef RGBLIGHT_ENABLE
         if (mode == 0) {
             rgblight_disable();
         } else {
             rgblight_enable();
         }
         rgblight_mode(mode);
+#endif
     }
     return 0;
 }
```

sekigon-gonnoc, 2021-08-01:
Why do you want to diable rgblights by compile options?
BLE Micro Pro intend to control this feature by configuration json files and use common firmware for all keyboards.
If you don't use any LED, then please set config.led.pin to 255 in your CONFIG.JSN

tadakado, 2021-08-01:
No strong reasons, just my rules.mk for AVR disabled the option.
Now I choose to set it in CONFIG.JSN.
Thanks!

closed 2021-08-01 by tadakado