sekigon-gonnoc/qmk_firmware #48: Pico-PIO-USB split questions.

hanetzer, 2022-01-29:
Heyo, ran into your work on the discord, had a few questions about your work.
1. which gpio pins are turned into USB using this?
2. if you use the usb pio for split, are both devices visible to the host computer (master and slave halves)?
3. if the answer to 2 is yes, do you think it would be possible to flash downstream devices while plugged in to upstream ones (instead of directly attached to the computer)
Thank you for your time, this looks very interesting.

sekigon-gonnoc, 2022-01-30:
1. Pin 0 is D+, and pin 1 is D-. You can change this by editing config https://github.com/sekigon-gonnoc/qmk_firmware/blob/rp2040/keyboards/pico_pico_usb/simple_split/simple_split.c#L35
2. No

hanetzer, 2022-01-30:
Ok, cool, and two more questions:

1. could one instead plug a non-qmk/rp2040 device into the pio-usb port (say for example, a bluetooth dongle or wireless mouse dongle) and have it work through the pico-pio-usb keyboard? Either directly thru the upstream device or down the line `computer->keyboard1->keyboard2->dongle`?
2. I'm assuming that if I flash this on a bare rpi pico the connection to the second one would be via the usb micro port on it? so `pico1[GP0,GP1]->pico2[usb-micro]`

sekigon-gonnoc, 2022-02-04:
1. Normal HID USB/Mouse probably recognized. For other devices, you have to write proper driver for those.
2. Yes
3. I selected them from my inventory. I'm assuming their properties are same

fxliang, 2023-01-31:
is it possible to connect two pico like bellow then keyboard which communicating via serial can also work in usb mode(connect with usb type-c wire)
pico1[GP0]   <---->   pico2[GP1]
pico1[GP1]   <---->   pico2[GP0]