sekigon-gonnoc/qmk_firmware #58: [Feature Request]  keyboard quantizer on Ultimate Hacking Keybaord (Trackpoint)

tcastelly, 2023-07-05:
## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Right now the quantizer works well with the ultimate hacking keyboard. But the extra modules like the Trackpoint Module does not work. 
Do you think it's something can be implemented?

Here the source code:
https://github.com/UltimateHackingKeyboard

sekigon-gonnoc, 2023-07-05:
Could you try increment `CFG_TUH_HID` in tusb_config.h?

https://github.com/sekigon-gonnoc/qmk_firmware/blob/quantizer_mini/tmk_core/protocol/pico/tusb_config.h#L135C39-L135C39

tcastelly, 2023-07-05:
Thank you for your answer. I'm a bit confuse sorry.

The [documentation](https://github.com/sekigon-gonnoc/keyboard-quantizer-doc/blob/master/mini/full.md) refers to the branch `keyboard/sekigon/keyboard_quantizer/mini-full`. In this branch I have the `keyboard_quantizer` folder but I don't have the `tmk_core/protocol/pico/tusb_config.h` file.

But in the branch `quantizer_mini` I have the expected file but I don't have the `keyboard_quantizer` folder. Have I to merge something?

I'm sorry I'm a bit lost.

sekigon-gonnoc, 2023-07-08:
https://github.com/sekigon-gonnoc/qmk_firmware/tree/quantizer_mini/keyboards/keyboard_quantizer/mini

sekigon-gonnoc, 2023-07-08:
I released new firmware with `CFG_TUH_HID` set to 8
https://github.com/sekigon-gonnoc/keyboard-quantizer-doc/releases/tag/mini-3

tcastelly, 2023-07-08:
It works amazing! Thank you so much!

closed 2023-07-08 by tcastelly